mit-acl/cadrl_ros #1: Is ford_msgs an open package on ROS?

yzxky, 2018-03-12:
I cannot find ford_msgs on ROS website, is it a customized package? Where can I get that package? Thx.

mfe7, 2018-03-12:
good point. that's actually a custom pkg i haven't uploaded to github yet.

i suggest you look into the cadrl_node.py file to see where those custom msgs are being used (i think clusters, plannermode, safeactions) and see if there is a way to modify the python code to handle the input your system current uses. for example, we have our laserscan clusters in a custom message format, but any existing message format you have to define other agents' positions, velocities etc could be fit into this sample code without too much trouble. that is probably much easier than trying to modify the rest of your system to use our custom message formats to work with this script.

i will try to remove those dependencies in the next few weeks

yzxky, 2018-03-13:
Thx. I am Kaiyuan Xu from CFINS, Tsinghua University, China. I have read your paper and think you have done a great job. Our lab is very interested in the topic you have introduced and are trying to do some extended research related to yours. We can get in touch and push forward the research of robot control in people-rich environment.

mfe7, 2018-05-27:
Sorry for the delay in response. Please find the `ford_msgs` repo at [this link](https://bitbucket.org/acl-swarm/ford_msgs/src) and use the `dev` branch. Let me know if that does not work and I will re-open the issue.

closed 2018-05-27 by mfe7

mudassirej, 2019-04-01:
Hi Mfe7. i also read your paper and i am trying to continue your work. i  want to know which simulation software you used for this project?

mfe7, 2019-04-02:
Thank you for the interest in the work. We wrote our own simulator for the
agent dynamics, hoping to release it at some point but no timeline yet.
This repo just contains the pre-trained network for inference, not the
training code.