mit-acl/cadrl_ros #2: About the training set.

chenzhiwo, 2018-06-07:
How can I obtain the initial training set that the paper say that is relased to the public?

mfe7, 2018-06-07:
Thanks for pointing this out. I have just uploaded the files and some instructions to a Dropbox folder: https://www.dropbox.com/sh/yu1spzhhj8c9akl/AAALo8yXSfQ1nxaU2KFRGWuTa?dl=0

I have also added this info to this repo's readme. Let me know if there are any questions.

closed 2018-06-07 by mfe7

chenzhiwo, 2018-06-08:
Here is a little problem, the `dataset_readme.txt` said that `states.shape ---> (100000, 26)`, however, for the regular dataset, `states.shape` should be `(100000, 28)`.

chenzhiwo, 2018-06-08:
Just a little confusing, thank you!

mfe7, 2018-06-08:
Take a look at the description of the state array for the two styles of
observations. One has 26 elements and the other has 28

Good point, I’ll try to make that more clear