mit-acl/cadrl_ros #11: The problem of running code

BingHan0458, 2020-12-05:
hello, author, I try to run 'ga3c_cadrl_demo.py' and it prints 'action: [1.2        0.52359878]'. I understand that it means the next action of the host_agent and is that correct?
And how to get the figures (Fig. 4 and Fig. 6) in you paper if I run the code 'network.py'?

BingHan0458, 2020-12-05:
And which models do network_01653000, network_01900000 and network_02360000 correspond to, respectively? How can I watch their structures?
Thanks!

mfe7, 2020-12-07:
Yes, you're correct that is the action for the host_agent. If there are multiple agents in the environment, they would each compute their next action separately (presumably before any of the agents execute the action at that timestep).

To re-create Figs 4 and 6 from the IROS paper, it may be easier to use our [gym environment repo](https://github.com/mit-acl/gym-collision-avoidance). In particular, there's a small test suite script that I think runs those test cases. This might also be a useful repo if you're looking to compare algorithms across random scenarios.

Whichever network file is loaded by default is the GA3C-CADRL-10 that should perform the best. The other ones are either trained for fewer steps or might have had different settings, so it's probably best to ignore the other network files.

closed 2020-12-07 by mfe7