mit-acl/cadrl_ros #12: from ford_msgs.msg import PedTrajVec, NNActions, PlannerMode, Clusters

BingHan0458, 2020-12-10:
Hey Michael,
Firstly, according to the README.md, I replaced 'git clone https://bitbucket.org/acl-swarm/ford_msgs.git' with 'git clone https://bitbucket.org/acl-swarm/ford_msgs.git -b dev' because I can't find the dev branch and run 'catkin_make'. It successed.
Then I tried to run the './cadrl_node.py', but I got an error 'ImportError: No module named ford_msgs.msg' in line 7, 'from ford_msgs.msg import PedTrajVec, NNActions, PlannerMode, Clusters'. 
So how can I do to solve the above two problems?
Thx. Kind regards.

BingHan0458, 2020-12-10:
And I want to know what is the corresponding python version of ROS? Python2.7 or Python3.X? When I run 'catkin_make', the python version displayed by default is python2.7. Will compilation errors occur later?

mfe7, 2020-12-10:
Great point - i think we merged that branch into master so you did the right thing.

Did you source the setup.bash associated with the catkin workspace containing ford_msgs? can you `rosmsg show` any of the ford msgs? that would point to whether the issue is with python or ros

That depends on which version of your OS and ROS you have installed -- eg this codebase was developed on a machine with ros-kinetic and ubuntu 16.04, which is designed for python 2.7. I think some of the newer ros versions support python3 but the python code in this repo is designed for python2.7.

HanBing0802, 2020-12-12:
Thank you very much. I tried your suggest 'source ~/catkin_ws/devel/setup.bash' and it success!
But I got another bug 'Unable to register with master node [http://localhost:11311]: master may not be running yet. Will keep trying.' when I run './cadrl_node.py'. So how can I do to solve it?
Thx. Kind regards.

luorui93, 2020-12-12:
Hi HanBing,
You need to have a roscore running first before you run the node.
Either from a roslaunch file or run roscore directly.

HanBing0802, 2021-01-03:
Hey Michael,
Thank you very much for your patience. During this period, I didn't continue to complete it due to personal reasons. Now there is a new problem in running the code './cadrl_node.py'. I look forward to your answer. The specific information of the bug is as follows:
`hello world from cadrl_node.py
2021-01-02 18:25:09.565110: I tensorflow/core/platform/cpu_feature_guard.cc:137] Your CPU supports instructions that this TensorFlow binary was not compiled to use: SSE4.1 SSE4.2 AVX AVX2 FMA
Traceback (most recent call last):
  File "./cadrl_node.py", line 596, in <module>
    run()
  File "./cadrl_node.py", line 585, in run
    pref_speed = rospy.get_param("~jackal_speed")
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/client.py", line 465, in get_param
    return _param_server[param_name] #MasterProxy does all the magic for us
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/msproxy.py", line 123, in __getitem__
    raise KeyError(key)
KeyError: '~jackal_speed'`
Thx. Kind regards.
Happy new year to you!
Bing Han

closed 2021-01-04 by mfe7