mit-acl/cadrl_ros #13: 'launch/cadrl_node.launch'

HanBing0802, 2021-01-03:
Hey Michael,
Thank you very much for your patience. Now there is a new problem in running the code 'launch/cadrl_node.launch'. I look forward to your answer. 

The specific information of the bug is as follows:
launch/cadrl_node.launch: line 1: syntax error near unexpected token `newline'
launch/cadrl_node.launch: line 1: `<launch>'

Thx. Kind regards.
Happy new year to you!
Bing Han

HanBing0802, 2021-01-03:
When I run by 'roslaunch cadrl_node.launch', it has a bug as follows:
... logging to /home/hb/.ros/log/a3201dbc-4d69-11eb-a154-000c291ba878/roslaunch-ubuntu-20219.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
started roslaunch server http://ubuntu:42785/
SUMMARY
========
PARAMETERS
 * /JA01/cadrl_node/jackal_speed: 1.2
 * /rosdistro: kinetic
 * /rosversion: 1.12.17
NODES
  /JA01/
    cadrl_node (cadrl_ros/cadrl_node.py)
ROS_MASTER_URI=http://localhost:11311
process[JA01/cadrl_node-1]: started with pid [20236]
hello world from cadrl_node.py
2021-01-03 03:50:40.976400: I tensorflow/core/platform/cpu_feature_guard.cc:137] Your CPU supports instructions that this TensorFlow binary was not compiled to use: SSE4.1 SSE4.2 AVX AVX2 FMA
('********\n*******\n*********\nJackal speed:', 1.2, '\n**********\n******')
Exception in thread Thread-5:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/timer.py", line 235, in run
    self._callback(TimerEvent(last_expected, last_real, current_expected, current_real, last_duration))
  File "/home/hb/catkin_ws/src/cadrl_ros/scripts/cadrl_node.py", line 389, in cbComputeActionGA3C
    predictions = self.nn.predict_p(obs, None)[0]
TypeError: predict_p() takes exactly 2 arguments (3 given)

mfe7, 2021-01-04:
are you using roslaunch to run that launch file? (that seems like a python error)

HanBing0802, 2021-01-04:
> are you using roslaunch to run that launch file? (that seems like a python error)

yes, I use 'roslaunch cadrl_node.launch' and the result is as follows:
`... logging to /home/hb/.ros/log/a3201dbc-4d69-11eb-a154-000c291ba878/roslaunch-ubuntu-20219.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
started roslaunch server http://ubuntu:42785/
SUMMARY
========
PARAMETERS
 * /JA01/cadrl_node/jackal_speed: 1.2
 * /rosdistro: kinetic
 * /rosversion: 1.12.17
NODES
  /JA01/
    cadrl_node (cadrl_ros/cadrl_node.py)
ROS_MASTER_URI=http://localhost:11311
process[JA01/cadrl_node-1]: started with pid [20236]
hello world from cadrl_node.py
2021-01-03 03:50:40.976400: I tensorflow/core/platform/cpu_feature_guard.cc:137] Your CPU supports instructions that this TensorFlow binary was not compiled to use: SSE4.1 SSE4.2 AVX AVX2 FMA
('********\n*******\n*********\nJackal speed:', 1.2, '\n**********\n******')
Exception in thread Thread-5:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/timer.py", line 235, in run
    self._callback(TimerEvent(last_expected, last_real, current_expected, current_real, last_duration))
  File "/home/hb/catkin_ws/src/cadrl_ros/scripts/cadrl_node.py", line 389, in cbComputeActionGA3C
    predictions = self.nn.predict_p(obs, None)[0]
TypeError: predict_p() takes exactly 2 arguments (3 given)`
I don't know why and I don't change anything in the code.

mfe7, 2021-01-05:
ah i think I changed the API for the neural network at some point and forgot to update the ros node. That line should work if you get rid of the 2nd argument to predict_p?

BingHan0458, 2021-01-05:
> ah i think I changed the API for the neural network at some point and forgot to update the ros node. That line should work if you get rid of the 2nd argument to predict_p?

yes, I have tried to get rid of the 2nd argument to predict_p and it doesn't have the previous bug but it does nothing. So I removed annotation ‘print "predictions", predictions’, then it loops infinitely and I don't know why it can loop and how to shutdown?

HanBing0802, 2021-01-05:
> ah i think I changed the API for the neural network at some point and forgot to update the ros node. That line should work if you get rid of the 2nd argument to predict_p?

emmm, I click ‘ctrl+c’ and it shuts down and the result is as follows.

^C[JA01/cadrl_node-1] killing on exit
[INFO] [1609811953.441362]: [/JA01/cadrl_node] Shutting down.
[INFO] [1609811953.442130]: Stopped JA01's velocity.
predictions: [0.087332   0.05659726 0.00574056 0.00810037 0.02762621 0.02491285
 0.12819767 0.11566862 0.06534816 0.21978024 0.26069608]
best action index: 10
shutting down processing monitor...
... shutting down processing monitor complete
done

Is this right? It need to stop manually instead of automatically by itself?

mfe7, 2021-01-05:
when you say it loops infinitely, is it publishing `Twist` commands? the ros node should run indefinitely and keep querying the NN at a fixed frequency

HanBing0802, 2021-01-05:
> when you say it loops infinitely, is it publishing `Twist` commands? the ros node should run indefinitely and keep querying the NN at a fixed frequency

I don't think it is publishing `Twist` commands because the print results are the same 
'predictions: [0.087332   0.05659726 0.00574056 0.00810037 0.02762621 0.02491285 0.12819767 0.11566862 0.06534816 0.21978024 0.26069608]
best action index: 10
'
and you say the ros node should run indefinitely and keep querying the NN at a fixed frequency, is that means I do right to shutdown manually by clicking ‘ctrl+c’ instead of automatically by itself？

mfe7, 2021-01-05:
what do you see when you rostopic echo the nn_cmd_vel topic?

sure, you can shut it down whenever you want by clicking ctrl+c, but i
would expect the vehicle command will stay the same as long as the input
observation is the same. so something needs to change on the input topics
to see a change in the selected action. normally i would start the ros
node, let it run while the robot moves around, then at the end of the
experiment ctrl+c the node.

HanBing0802, 2021-01-05:
> what do you see when you rostopic echo the nn_cmd_vel topic? sure, you can shut it down whenever you want by clicking ctrl+c, but i would expect the vehicle command will stay the same as long as the input observation is the same. so something needs to change on the input topics to see a change in the selected action. normally i would start the ros node, let it run while the robot moves around, then at the end of the experiment ctrl+c the node.
> […](#)
> -- Michael Everett Postdoctoral Associate Department of Aeronautics and Astronautics Massachusetts Institute of Technology (MIT) mfe.mit.edu

when I run 'rostopic echo /nn_cmd_vel', it has a bug as follows:
WARNING: topic [/nn_cmd_vel] does not appear to be published yet

mfe7, 2021-01-06:
i think the topics are put under the /JA01 namespace by default in that launch file. so it would probably be `rostopic echo /JA01/nn_cmd_vel` -- but you can do `rostopic list` to see what topics are being published

HanBing0802, 2021-01-06:
> i think the topics are put under the /JA01 namespace by default in that launch file. so it would probably be `rostopic echo /JA01/nn_cmd_vel` -- but you can do `rostopic list` to see what topics are being published

I am a beginner, so there are many problems, and thank you for your patience,
Yes, I try `rostopic echo /JA01/nn_cmd_vel` but it also loops infinitely and prints the same results
"linear: 
  x: 0.0
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0"
I think that means the agent doesn't move anymore. And then I run 'rosrun rqt_graph rqt_graph' and there are only a node '/JA01/cadrl_node' without any activate topics. So it's the final results? But I indeed don't know why they loop infinitely not only 'roslaunch cadrl_node.launch' but also 'rostopic echo /JA01/nn_cmd_vel'?

mfe7, 2021-01-07:
That seems right to me initially. The command should be to stay in place while waiting for a goal. Have you set a goal (eg by publishing a msg to the topic corresponding to the goal)? Once it receives a goal, the agent should spin in place (nonzero angular z) until aligning with the goal, then will move forward according to the NN output.

I think rqt_graph only shows topics that something is subscribing to

It loops infinitely because this ROS node is meant for a robot that continually receives new sensor data, and thus observations of agent states. If you're just looking for the policy on its own (eg for a single observation timestep, query the NN), there's a separate repo that would be simpler to use and not require ROS.

HanBing0802, 2021-01-08:
Hey Michael,
You means that if I just want to know the path of the robot to avoid obstacles, I need to publish a msg to set a goal position. Is that true? If so, I can understand why the output of `rostopic echo /JA01/nn_cmd_vel` is zero. Before this I always thought that the robot's path could be visualized after `roslaunch cadrl_node.launch` because I saw the function 'visualize_**()'. I'm sorry, because I didn't study ROS before, I took it for granted. I think next I should study ROS such as how to publish a msg.
Thank you very much!

mfe7, 2021-01-10:
That is correct -- the cadrl node needs to know the goal before it can compute velocities that will get the robot to that goal. I'll close this issue for now but feel free to re-open if other questions come up.

closed 2021-01-10 by mfe7