mit-acl/cadrl_ros #15: Generate observation data from sensors

TehamC, 2021-01-04:
Hi Michael, 
I am working with cadrl for a few weeks now and I want to thank you for your amazing work.
Currently my turtlebot can navigate on a map while avoiding dynamic obstacles. (Basically following a global plan by sending subgaosl, similar to the youtube video: https://www.youtube.com/watch?v=CK1szio7PyA )

In the simulation I directly publish the Cluster msgs from the odom data of the obstacles. However I would like to generate them from sensor data for example the local costmap using the move base package. 
So I wanted to know if you can give me a hint on how you generated the cluster msgs in ur example.

In Minute 1:60 you can see the jackal waiting for the lady to finish her order, meaning it is aware of the pillar in front of him. Are static obstacles from the map also published as cluster msgs ?

thanks in advance :)

mfe7, 2021-01-05:
That's exciting to hear you've implemented it on hardware!

In that video (from the IROS 17 paper), the main way we handled static obstacles (such as the pillar) was that we start with a set of motion primitives (essentially straight lines in many directions from the vehicle center), do collision checking on them given the current costmap (trim each MP at the closest occupied cell in the costmap), then feed a version of those trimmed MPs to the cadrl node (NNActions msg). Each NNActions msg essentially contains the distance and angle that the vehicle could travel in a straight line safely wrt static obstacles. Then we would sample several points along each of the trimmed MPs according to the NN value fn to see which is the best action to take wrt dynamic obstacles. I think this process is described a little more in my master's thesis.

This codebase implements the IROS 18 paper, and I don't remember if this repo implements that idea described above. If nothing else, you could probably pass static obstacles in as ClusterMsgs with zero velocity (as you mentioned) and the NN policy should be able to avoid them, since there were some static agents seen during training. To get those clusters right away from sensor data might require a bit of work -- what we had done was cluster the 2D lidar data and track those clusters over time (using a dynamic k-means algorithm), so some of those clusters' velocities might happen to be zero.

TehamC, 2021-01-08:
Ok these are some very interesting info,
what I am concerned of this approach is the efficiency. Does cadrl support other agent shapes than circles ?

[image]

In this example I could approximate the rectangular shaped obstacle using 2 circular agents or one oval shaped agent. If only the circle approach is possible then the computation time may increase by a lot since there could be a lot of obstacles inside a map.

mfe7, 2021-01-10:
In its current form the trained policy only supports circular obstacles. So you might be able to get away with the combination of circles. I suspect the computation time of the policy won't be too bad as it will just require additional LSTM cycles. But I'm curious to hear how it turns out if you go that route.

TehamC, 2021-01-16:
Okay thanks alot for ur advice, 
U were right, the performance impact is actually neglible. I just manually added the static map as agents with zero velocity and it works really well. :)

closed 2021-01-16 by TehamC

fridayfang, 2022-05-23:
> Hi Michael, I am working with cadrl for a few weeks now and I want to thank you for your amazing work. Currently my turtlebot can navigate on a map while avoiding dynamic obstacles. (Basically following a global plan by sending subgaosl, similar to the youtube video: https://www.youtube.com/watch?v=CK1szio7PyA )
> 
> In the simulation I directly publish the Cluster msgs from the odom data of the obstacles. However I would like to generate them from sensor data for example the local costmap using the move base package. So I wanted to know if you can give me a hint on how you generated the cluster msgs in ur example.
> 
> In Minute 1:16 you can see the jackal waiting for the lady to finish her order, meaning it is aware of the pillar in front of him. Are static obstacles from the map also published as cluster msgs ?
> 
> thanks in advance :)

hello, I'm also working with cadrl on turtlebot3. But I doubt it really work well on turtlebot3.
As far as I know, max speed of turtlebot3 is 0.26m/s;and for the network it's 1.2m/s
Can you help me? How to migrate it to turtlebot3 robot or share me some code of your previous? This is my email 1569496890@qq.com
Many thanks for you.

fridayfang, 2022-05-23:
@TehamC

mfe7, 2022-05-23:
I think there's a max robot speed parameter in the ROS node, which should get passed to the NN when querying the policy. The policy was trained with randomly sampled max speeds (can't remember the range exactly, but it should be in the paper), so it may actually work for this lower speed as well.