mit-acl/cadrl_ros #17: Can it run on Ubuntu 20.04 with noetic?

LukeZhi, 2021-12-21:
Can it run on Ubuntu 20.04 with noetic?

mfe7, 2021-12-21:
i haven't tried it but nothing jumps out of why it wouldn't work

closed 2021-12-21 by mfe7

mfe7, 2021-12-23:
I just made a small commit changing some print statements so that the node starts up with python3+noetic

LukeZhi, 2021-12-23:
Thanks, but I think there may be still some errors or not...
In Ubuntu20.04, the default python is pytho3.8 and the tensorflow I can install  only is tensorflow2.x. So there are not placeholder. Even I use "import tensorflow.compat.v1 as tf"  there still some errors such as "has no attribute 'contrib' " and : 
"TypeError: Dimension value must be integer or None or have an __index__ method, got TensorShape([Dimension(None), Dimension(256)])"
I don't know where I make mistake with your code.

mfe7, 2021-12-23:
Oh I see. I'm running Python 3.6.9 here rather than the default 3.8 that comes with ubuntu. And I haven't sent commands to the robot that would cause the NN to be queried. I'm not sure I'll have time to get it working with tensorflow2.x -- so one possible path forward would be to run this node in a virtualenv with an older python version that supports tf1 (I have enjoyed using pyenv to manage python versions). I'll post any updates here if I get a chance to update the compatibility.