mit-acl/cadrl_ros #19: Sir, I have some questions about CADRL.

CAI23sbP, 2022-07-07:
I have two questions.
(1) Fundamentally, is it possible to avoid CADRL by reflecting newly updated obstacle information that is not included in the static map or dynamic objects other than people?

(2) If i have trained 4 agents environment , CADRL could act well 5 or more agents environment?

mfe7, 2022-07-08:
I'm not following what you mean in (1). For (2), please refer to our papers - the short answer is yes, the approaches can scale to larger numbers of agents than were seen during training.

CAI23sbP, 2022-07-08:
Thank you for answer about these question, sir.
The first question mean is..
(1)Assuming that originally a obstacle had been static but for any reason the obstacle was slided, then CADRL could avoid that?

CAI23sbP, 2022-07-08:
> I'm not following what you mean in (1). For (2), please refer to our papers - the short answer is yes, the approaches can scale to larger numbers of agents than were seen during training.

Thank you for answer about these question, sir.
The first question mean is..
(1)Assuming that originally a obstacle had been static but for any reason the obstacle was slided, then CADRL could avoid that?

mfe7, 2022-07-12:
Sure, if the velocity estimate of the obstacle (part of the observation vector) changes from zero to something else, I would still expect CADRL to be able to avoid the obstacle.

closed 2022-07-12 by mfe7