mit-acl/cadrl_ros #22: preset_testCases

TUzizi, 2022-10-11:
Hi, sorry to bother you again.

I find the test cases (**preset_testCases**, num_agents:10) you used. I try to run again using the checkpoint file([gym-collision-avoidance](https://github.com/mit-acl/gym-collision-avoidance)/[gym_collision_avoidance](https://github.com/mit-acl/gym-collision-avoidance/tree/release/gym_collision_avoidance)/[envs](https://github.com/mit-acl/gym-collision-avoidance/tree/release/gym_collision_avoidance/envs)/[policies](https://github.com/mit-acl/gym-collision-avoidance/tree/release/gym_collision_avoidance/envs/policies)/[GA3C_CADRL](https://github.com/mit-acl/gym-collision-avoidance/tree/release/gym_collision_avoidance/envs/policies/GA3C_CADRL)/[checkpoints](https://github.com/mit-acl/gym-collision-avoidance/tree/release/gym_collision_avoidance/envs/policies/GA3C_CADRL/checkpoints)/IROS18/network_01900000) you provided with vpref=1.0, DT=0.2. But it fails. I try to reduce the radius of the agent from 0.5 to 0.3, then it works but with a slightly different trajectory.
[image]

[image]
[image]

Is this normal or am I using the wrong parameter file?
Thanks for your help~

mfe7, 2022-10-11:
Does it work with dt=0.1? I recall that we used dt=0.2 in training to expedite the process but dt=0.1 in evaluation

TUzizi, 2022-10-11:
No. I tried 0.1 and 0.01, the agent moves slowly but has the same trajectory as 0.2. am I using the wrong parameter file?
(Why do you use different DT in training and evaluation? )

mfe7, 2022-10-12:
Hmm that does seem like the right checkpoint file. I can try to run this again locally and let you know if I see the same thing.

We used a smaller dt in evaluation to match what we were running on the hardware implementation for real-time performance, but we used a larger dt in training because the computation time was prohibitive to go much smaller back several years ago when we were training these policies.

mfe7, 2022-10-13:
I'm running the `SmallTestSuite` config with 2 agents locally. I'm able to replicate the SA-CADRL trajectories, but there must be something that's changed with the GA3C-CADRL-10 policy since generating those figures, because I am seeing different behavior even for the simple 2-agent scenarios. I can try to spend a little more time looking around, but that code has undergone a lot of changes, so I'm not sure how easy it will be to track down.

I also confirmed in a really old other repo that we were indeed using radius = 0.5 for that 10-agent circle case you tried.

TUzizi, 2022-10-13:
Thank you for your help ☺. If you succeed, could you please share your test code with me, so I can know why I failed?

mfe7, 2022-10-17:
Hi - I spent some time digging up and running old versions of the code (from 2018) and am not having any success. My current hypothesis is that there might be a long lost checkpoint file that was used in generating the results, but accidentally overwritten sometime later on.