mit-acl/cadrl_ros #24: Question regarding pose information

TomasMendozaHN, 2023-05-11:
Hello :) 
First of all, great job! It's absolutely amazing what you all did for this research. 
As a total beginner in ROS, I do want to ask two things:

1. After building your environment, how do I run it? 
2. At any stage you run SLAM? I am wondering since I saw on the youtube video that SLAM was running...or, because you are suscribed to the "PoseStamped" message then you are able to obtain the global coordinates without having to explicitely run SLAM?

Again, I am just beginning to ROS and would love to understand it a bit better :) Thank you!

mfe7, 2023-05-11:
After building the package, you can run the cadrl node using our launch file (e.g., `roslaunch cadrl_ros cadrl_node.launch`). You will probably need to remap some topics based on the rest of your autonomy stack.

When we deploy this on the robot, we do have a separate node that estimates the vehicle's pose. We typically would map the environment once with a SLAM package, then just use AMCL to localize in that map for future experiments and testing of the planning algorithms. The PoseStamped msg you refer to would be one of the outputs of the localization node.

closed 2023-05-19 by mfe7

TomasMendozaHN, 2023-05-22:
Thank you very much!!! :)