mit-acl/cadrl_ros #26: How to consider map?

CAI23sbP, 2023-10-20:
How are you @mfe7?
I have found "Robot designed for socially acceptable navigation", and i read it. 
I have a single question about diffusion map in page 48.
-> How to merge diffusion map and CADRL?
Thank you for reading this issue.

mfe7, 2023-10-22:
the way we used diffusion map and cadrl together in that thesis (from 2017) is: build a 2d occupancy map of the environment ahead of time, then online: given a global goal coordinate, repeatedly use the diffusion map algorithm to choose a subgoal 5-10m ahead of the robot's position that should lead toward the global goal (accounting for static obstacles in the prior map), then pass the current subgoal to CADRL to select a velocity vector (considering the dynamic obstacles in the present environment).

CAI23sbP, 2023-10-23:
Thank you for your reply

closed 2023-10-23 by CAI23sbP