mit-acl/cadrl_ros #27: problem about using cadrl_ros in pedsim_ros environment

JiangxinkeTao, 2025-04-13:
I hope this message finds you well. I have set up a dynamic pedestrian environment using pedsim_ros (https://github.com/srl-freiburg/pedsim_ros) and utilized the training network (checkpoints) you provided. However, the navigation performance in the dynamic environment is not ideal. I would like to know if it's necessary to retrain the network, or if there are any pre-trained networks available for direct use. Thank you for your guidance!

mfe7, 2025-04-15:
which agent is using the cadrl policy?

the format of the input vector would be my first place to look, as this contains some important tunable parameters such as the two agents' radii, and then the coordinate frames of the agent's goal and other agent's velocities is also important to get right. 

also, the cadrl_ros policy doesn't consider the static walls, so it should avoid the other agent but treats everything else as open space. to handle the static walls, since this cadrl policy has a discrete action space, you could consider assigning zero probability to the actions (v_x, \omega) that will drive into a wall, then select the argmax of the remaining actions -- this hack may lead to weird behavior since it differs from how the policy was trained, but at least would keep it from going into walls

JiangxinkeTao, 2025-04-16:
Thank you very much for your reply!
The radius of the agents is 0.3m, and although the robot is square for visualization, its effective radius is also 0.3m in practical applications. The speed, position, local target goal, and dynamic obstacle position information of the agents are all set according to the requirements in the Readme.md. I have printed the relevant information in the terminal, and there are no errors in the information.
Although the walls are displayed, they do not participate in the navigation process; they are only used for visualization.

JiangxinkeTao, 2025-04-16:
The `trans_pkg` is used to transform the data type based on pedsim to the type based on cadrl_ros. 

[trans_pkg.zip](https://github.com/user-attachments/files/19772247/trans_pkg.zip)

[image]